ros-industrial/kuka_experimental #182: Now able to establish connection between KR_C4 and ROS.

ranjitkathiriya, 2021-02-25:
Hey there,

I am trying to establish a connection between the RSI interface and KUKA KR_20 R1810-2, a KR_C4 controller.

Up till now, I have followed below instructions:

1. [Configuring RSI on the controller.](https://github.com/ros-industrial/kuka_experimental/tree/indigo-devel/kuka_rsi_hw_interface/krl/KR_C4)

2. [Problem to set-up RSI interface with KR6 R900 and this answer.](https://github.com/ros-industrial/kuka_experimental/issues/115#issuecomment-371521574)

I have set the Ip Address is shown below:

Smart-pad Ip address:
Windows 10 OS in a smart pad: 192.168.0.1
Not changed, and not safe to do so.
Smartpad virtual5 KLI: 172.31.1.147
Virtual 6: 192.168.1.121 
Windows interface is selected.
Virtual 7 RSI: 192.168.2.121 
Linux PC:
192.168.1.210   windows interface
192.168.2.210	  RSI

Selecting User as Administrator in the smart pad and after Minimize HMI (Start-up > Service > Minimize HMI), I window 10 of smart pad I can ping to 192.168.1.121, 192.168.2.121, and 192.168.1.210, but I am not able to ping 192.168.2.210. On the other side, I can ping all the smart-pad addresses, e.g. (172.31.1.147, 192.168.1.121, 192.168.2.121 ).

In ros_rsi_ethernet.xml, I have tried adding both IP addresses of my Linux pc 192.168.2.210 and 192.169.1.210.

In test_params.yml, I have changed the IP address to 192.168.2.210. Further, I have added the below line just before kuks_hardware_interface.

```
<arg name="model" default="$(find kuka_kr16_support)/urdf/kr16_2.xacro"/>
<param name="robot_description" command="$(find xacro)/xacro $(arg model)" />
```

This is the output I am getting :

```
kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~/ws_moveit$ roslaunch kuka_rsi_hw_interface test_hardware_interface.launch sim:=false
... logging to /home/kuka-ai/.ros/log/63f99866-7771-11eb-bd0a-efc131f6c940/roslaunch-kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU-8082.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:40857/

SUMMARY
========

PARAMETERS
 * /controller_joint_names: ['joint_a1', 'joi...
 * /joint_state_controller/publish_rate: 50
 * /joint_state_controller/type: joint_state_contr...
 * /position_trajectory_controller/action_monitor_rate: 20
 * /position_trajectory_controller/joints: ['joint_a1', 'joi...
 * /position_trajectory_controller/state_publish_rate: 50
 * /position_trajectory_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.9
 * /rsi/listen_address: 192.168.1.210
 * /rsi/listen_port: 49152

NODES
  /
    controller_spawner (controller_manager/spawner)
    kuka_hardware_interface (kuka_rsi_hw_interface/kuka_hardware_interface_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

ROS_MASTER_URI=http://localhost:11311

process[kuka_hardware_interface-1]: started with pid [8098]
process[controller_spawner-2]: started with pid [8099]
[ INFO] [1614265728.636066787]: Starting hardware interface...
process[robot_state_publisher-3]: started with pid [8101]
[ INFO] [1614265728.647552308]: Loaded kuka_rsi_hardware_interface
[ INFO] [1614265728.650771039]: Setting up RSI server on: (192.168.1.210, 49152)
[ INFO] [1614265728.675373482]: Waiting for robot!
[WARN] [1614265728.882024]: DEPRECATION warning: --shutdown-timeout has no effect.
[INFO] [1614265728.883432]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1614265728.887354]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1614265728.891932]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1614265728.896652]: Loading controller: position_trajectory_controller
[INFO] [1614265728.924515]: Loading controller: joint_state_controller
[INFO] [1614265728.929037]: Controller Spawner: Loaded controllers: position_trajectory_controller, joint_state_controller
```

On the controller side I am getting this error: 

[image]
[image]

My configuration is correct? 
And why am I not able to communicate?

Thanks in advance for your help.

destogl, 2021-03-01:
On a first glance your RSI listening address is set wrong in ROS:

/rsi/listen_address: 192.168.1.210

Shouldn't this be 2.210?

ranjitkathiriya, 2021-03-01:
Yes, first, I have tried 192.168.2.210(RSI subnet mask) and got the same error.

ranjitkathiriya, 2021-03-08:
Thanks, @destogl, 
for replying to my answer and I have tried with both IP addresses but I am not able to solve.

I did all things from the scratch again, and I found few points, but I am confused that it a true or wrong.

1. I was using the kuka_experimental library from GitHub for ROS configuration, In that, I am using the kuka_kr16_support package for the robot, but my robot is KR20 R1810-2. I have checked the difference between both robots, and I found that some links given in both pictures are different.

Again: I am not sure that this causes the problem or not?

Simulator:

[image]

Real Robot:

[image]

I think that in the robot some links look different(link 1 and link 3); that may be an issue for ROS connection or not?

2. Whenever I am running my program:

I see this type of behavior: First, it will execute PTP {A1 0, A2 -90, A3 90, A4 0, A5 90, A6 0} this line, and release the play button and play it again then my cursor stops here in HALT. I have checked my file is in the proper location, and I think that this condition is not satisfying

ret = RSI_CREATE("ros_rsi.rsi",CONTID,TRUE)

IF (ret <> RSIOK) THEN

HALT # Over here

ENDIF

3. After Halt when I press the play button again, I am getting two warnings:

1. RSI: cannot set any outputs due to operator protection

Originator:KS

(Do we have to turn some operator flag on for this?)

2. RSI_MOVECORR: Motion active

Originator:0

And in Auto mode: all things up till now remain the same but I get this extra warning.

3. Process active

KSS01372

Thanks for helping

ranjitkathiriya, 2021-03-16:
It is a controller problem.

gavanderhoorn, 2021-03-16:
> It is a controller problem because it was not able to find ros_rsi.rsi file so the controller was giving a flag of 27

but aren't you supposed to copy that file to the controller as part of the setup procedure?

ranjitkathiriya, 2021-03-24:
Hello @gavanderhoorn , @destogl ,

Sorry for my previous message, this issue was of my

closed 2021-03-24 by ranjitkathiriya

SaeedRT93, 2025-01-23:
Hi @ranjitkathiriya ,

Didi you manage to creat the files for the newer version of RSI 4+?