ros-industrial/kuka_experimental #183: Ros Noetic support

ranjitkathiriya, 2021-03-10:
Does the kuka_experimental package work fine with Ros noetic and foxy?

gavanderhoorn, 2021-03-10:
> Does the kuka_experimental package work fine with Ros noetic

I've not seen problems reported about it not building, so I assume it does.

> and foxy?

that's a ROS 2 version. So: no.

You could perhaps take a look at [dignakov/ros2_control_demo_drivers](https://github.com/dignakov/ros2_control_demo_drivers), but note that's essentially a completely different package.

gavanderhoorn, 2021-03-10:
As this is not an issue with any of the packages in this repository, I'm going to close it.

Feel free to keep commenting of course.

closed 2021-03-10 by gavanderhoorn

ranjitkathiriya, 2021-03-10:
Thanks for solving my query. 

:-)