ros-industrial/kuka_experimental #187: Continues Movement

MBaranPeker, 2021-03-22:
Hi @gavanderhoorn , I have a question for a application similar to jogging. Robot start at position A to move B while that it calculate waypoints then send it robot then the robot execute the action then stop untill next  waypoints to execute. What I want is while robot executing the first action(between A to B), I want to sent robot to the waypoints between position B and C then robot can add up the waypoints for continues movement. I do wonder is it possible to do that without changing the velocity.

In Kuka Work Visual Software you can do that with CONT command but I am not sure do EKI or RSI  interface will let us to do that ? Do we need to update in the interface for CONT command ?

Thanks,

Position A = (0, 0, 0) Position B = (0, 0, 10) Position C = (0, 0, 20) Position D = (10, 0, 20) .

[image]

gavanderhoorn, 2021-03-30:
> In Kuka Work Visual Software you can do that with CONT command but I am not sure do EKI or RSI interface will let us to do that ? Do we need to update in the interface for CONT command ?

RSI is a low-level position control interface which moves the robot according to whatever commands you sent it (provided they are feasible, ie: within kinematics and dynamic limits).

Whether a motion is "smooth" depends entirely on whatever/whoever generates the position set points (ie: something like a trajectory generator).

What you describe sounds more like the base interpolation between poses as the controller would perform when programming using KRL (I believe that's what you mean when you refer to Work Visual).

RSI and KRL use a different approach, where the latter is somewhat higher level than the former and RSI will require you to take care of more details.

Summarising: there is "nothing" immediately in RSI which will smooth out motions between destinations, that's your responsibility.

(I put "nothing" In quotes as there are filter blocks in RSI which *could* be used, but those are not in the default configuration used by `kuka_rsi_hw_interface`)

gavanderhoorn, 2021-03-30:
At a high-level, what you describe is of course also how RSI works: if A, B, C, and D would be 4 ms (in time) apart and their distance in space would (inversely via time) lead to motions within the kinematic and dynamic limits.

That doesn't really help you, but in general, robot controllers interpolate between poses. RSI just makes the controller interpolate between poses at a high time resolution.

gavanderhoorn, 2021-03-30:
As this is not an issue with the packages in this repository, I'm going to close it.

Feel free to keep commenting.

closed 2021-03-30 by gavanderhoorn