ros-industrial/kuka_experimental #191: KR10r1420 support file

MBaranPeker, 2021-05-24:
Hi,
I am trying to create a support file for kuka kr10r1420 robot. Unfortunately when I try to run on Rviz links moves separately. when I move joint_1 all links moves together as it should be. When I move joint_2 except link_1 other links move. I could not figure out what is the problem with that. I will appreciate if anybody could help. 
Additional Informations ; 

I set link information same as datasheet.(Checked twice)
In xacro file i set <origin xyz="0 0 0" rpy="0 0 0"/> for each link. Because I don't know how to calculate it. 
Collision Links same as visual links I didn't crate new collision cad files. 
while creating new support file I reference kr10r1100sixx robot.

Original Link locations in Blender.

[image]

Rviz

[image]
[image]

tingelst, 2021-05-24:
You need to move the origin of each mesh to (0,0,0).

MBaranPeker, 2021-05-24:
@tingelst do you mean for each joint origin should be (0,0,0) in stl files?

tingelst, 2021-05-24:
> @tingelst do you mean for each joint origin should be (0,0,0) in stl files?

Yes

MBaranPeker, 2021-05-24:
@tingelst Thanks.

MBaranPeker, 2021-05-25:
@tingelst 
Hi, 
This is the result If I make orijin of each joint (0,0,0) in stl files.(Geometry to orijin)
[image]
This is how I did that. (Geometry to orijin)
[image]

Another experiment result same as yesterday.(Orijin to Geometry)
[image]
 
How I did that.(Orijin to Geometry)

[image]

MBaranPeker, 2021-05-25:
I think I found the problem.

After setting all link origins to (0,0,0) (geometry to orijin) and updating Urdf link orijins to orijin to geometry values in blender. Links locations looks fine in Rviz.

    <link name="${prefix}link_1">
      <visual>
        <origin xyz="0.050365  -0.004451 0.356414" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://kuka_kr10_support/kr10r1420/meshes/kr10r1420/visual/link_1.stl" />
        </geometry>
        <xacro:material_kuka_orange />
      </visual>
      <collision>
        <origin xyz="0.050365  -0.004451 0.356414" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://kuka_kr10_support/kr10r1420/meshes/kr10r1420/collision/link_1.stl" />
        </geometry>
      </collision>
<inertial>
            <mass value="0.1" />
            <inertia ixx="0.03" iyy="0.03" izz="0.03" ixy="0.0" ixz="0.0" iyz="0.0" />
        </inertial>
    </link>

When I update the URDF file I set orijin of each joint (0,0,0)(Because I don't know how to find the correct values). I guess I need to set joint orijins. What is the best way to find that values? Thanks.

    <joint name="${prefix}joint_a1" type="revolute">
      <origin xyz="0 0 0" rpy="0 0 0"/>
      <parent link="${prefix}base_link"/>
      <child link="${prefix}link_1"/>
      <axis xyz="0 0 -1"/>
      <limit effort="0" lower="${-DEG2RAD * 170}" upper="${DEG2RAD * 170}" velocity="${DEG2RAD * 220}"/>
    </joint>

MBaranPeker, 2021-05-25:
<xacro:macro name="kr10r1420" params="prefix">
    <link name="${prefix}base_link">
      <visual>
        <origin xyz="0 0 0" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://kuka_kr10_support/kr10r1420/meshes/kr10r1420/visual/base_link.stl" />
        </geometry>
        <xacro:material_kuka_black />
      </visual>
      <collision>
        <origin xyz="0 0 0" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://kuka_kr10_support/kr10r1420/meshes/kr10r1420/collision/base_link.stl" />
        </geometry>
      </collision>
        <inertial>
            <mass value="0.1" />
            <inertia ixx="0.03" iyy="0.03" izz="0.03" ixy="0.0" ixz="0.0" iyz="0.0" />
        </inertial>
    </link>
    <link name="${prefix}link_1">
      <visual>
        <origin xyz="0.050365  -0.004451 0.356414" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://kuka_kr10_support/kr10r1420/meshes/kr10r1420/visual/link_1.stl" />
        </geometry>
        <xacro:material_kuka_orange />
      </visual>
      <collision>
        <origin xyz="0.050365  -0.004451 0.356414" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://kuka_kr10_support/kr10r1420/meshes/kr10r1420/collision/link_1.stl" />
        </geometry>
      </collision>
<inertial>
            <mass value="0.1" />
            <inertia ixx="0.03" iyy="0.03" izz="0.03" ixy="0.0" ixz="0.0" iyz="0.0" />
        </inertial>
    </link>
    <link name="${prefix}link_2">
      <visual>
        <origin xyz="0.143578 -0.119954 0.745506" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://kuka_kr10_support/kr10r1420/meshes/kr10r1420/visual/link_2.stl"/>
        </geometry>
        <xacro:material_kuka_orange />
      </visual>
      <collision>
        <origin xyz="0.143578 -0.119954 0.745506" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://kuka_kr10_support/kr10r1420/meshes/kr10r1420/collision/link_2.stl" />
        </geometry>
      </collision>
<inertial>
            <mass value="0.1" />
            <inertia ixx="0.03" iyy="0.03" izz="0.03" ixy="0.0" ixz="0.0" iyz="0.0" />
        </inertial>
    </link>
    <link name="${prefix}link_3">
      <visual>
        <origin xyz="0.284416 -0.005239 1.06527" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://kuka_kr10_support/kr10r1420/meshes/kr10r1420/visual/link_3.stl" />
        </geometry>
        <xacro:material_kuka_orange />
      </visual>
      <collision>
        <origin xyz="0.284416 -0.005239 1.06527" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://kuka_kr10_support/kr10r1420/meshes/kr10r1420/collision/link_3.stl" />
        </geometry>
      </collision>
<inertial>
            <mass value="0.1" />
            <inertia ixx="0.03" iyy="0.03" izz="0.03" ixy="0.0" ixz="0.0" iyz="0.0" />
        </inertial>
    </link>
    <link name="${prefix}link_4">
      <visual>
        <origin xyz="0.697026 -0.000196 1.08533" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://kuka_kr10_support/kr10r1420/meshes/kr10r1420/visual/link_4.stl" />
        </geometry>
        <xacro:material_kuka_orange />
      </visual>
      <collision>
        <origin xyz="0.697026 -0.000196 1.08533" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://kuka_kr10_support/kr10r1420/meshes/kr10r1420/collision/link_4.stl" />
        </geometry>
      </collision>
<inertial>
            <mass value="0.1" />
            <inertia ixx="0.03" iyy="0.03" izz="0.03" ixy="0.0" ixz="0.0" iyz="0.0" />
        </inertial>
    </link>
    <link name="${prefix}link_5">
      <visual>
        <origin xyz="0.8262 0.004332 1.08334" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://kuka_kr10_support/kr10r1420/meshes/kr10r1420/visual/link_5.stl" />
        </geometry>
        <xacro:material_kuka_orange />
      </visual>
      <collision>
        <origin xyz="0.8262 0.004332 1.08334" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://kuka_kr10_support/kr10r1420/meshes/kr10r1420/collision/link_5.stl" />
        </geometry>
      </collision>
<inertial>
            <mass value="0.1" />
            <inertia ixx="0.03" iyy="0.03" izz="0.03" ixy="0.0" ixz="0.0" iyz="0.0" />
        </inertial>
    </link>
    <link name="${prefix}link_6">
      <visual>
        <origin xyz="0.886629 -0.000014 1.08052" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://kuka_kr10_support/kr10r1420/meshes/kr10r1420/visual/link_6.stl" />
        </geometry>
        <xacro:material_kuka_pedestal />
      </visual>
      <collision>
        <origin xyz="0.886629 -0.000014 1.08052" rpy="0 0 0"/>
        <geometry>
          <mesh filename="package://kuka_kr10_support/kr10r1420/meshes/kr10r1420/collision/link_6.stl" />
        </geometry>
      </collision>
<inertial>
            <mass value="0.1" />
            <inertia ixx="0.03" iyy="0.03" izz="0.03" ixy="0.0" ixz="0.0" iyz="0.0" />
        </inertial>
    </link>
    
    <link name="${prefix}tool0"/>

gavanderhoorn, 2021-05-25:
You cannot use *geometry to origin*, as that will place the origin of the mesh in its centre of mass (ie: in the middle of "all the pixels/faces/vertices"). That's not what @tingelst meant to say.

The origins of the meshes should be in the same location as the origins of the joints *in the real robot*.

So for `base_link`, its origin should be at the bottom of the link, right where the `joint_a1` axis intersects with the bottom face of mesh.

It may be helpful to open some of the meshes of the other models. That may give you more of an idea.

Once you get that sorted out, you'll want to set the `origin` elements of each `visual` to `0 0 0`, as they then don't need any additional transformation to be rendered in the correct place.

Again: looking at some of the existing models and their meshes may be informative.

gavanderhoorn, 2022-03-10:
@MBaranPeker: did you get things to work?

If yes: would you be willing to contribute your variant to the `kuka_kr10_support` package?

aatb-ch, 2022-03-10:
I am actually setting up RSI to work with a KR10 R1420 so would be very much willing to help on this, if this could be a good first PR, if @MBaranPeker can provide anything I could support or help out, otherwise I will have something in the next couple of weeks which I could submit.

MBaranPeker, 2022-03-11:
@gavanderhoorn  No, I had another robot to use and time was short, so that I couldn't continue developing support file for KR10-R1420.
@aatb-ch thank you, It would be perfect for the community.

aatb-ch, 2022-03-11:
Alright no problem, would you mind sharing your existing files? I'm fine starting over from scratch but it could be helpful references.

MBaranPeker, 2022-03-11:
@aatb-ch I do have stl files( had some problems). I can share them with you. But I could not find urdf files( also had some problems ).  I suggest you to download it from KUKA website to be sure.

aatb-ch, 2022-03-11:
Sure, understood. Will start clean then, thank you.

aatb-ch, 2022-03-28:
submitted PR #216