ros-industrial/kuka_experimental #192: Support for Kr6 R7002 robot

uthiralakshmi, 2021-06-10:
Hey hi,

I was just hoping to get details about ROS interface with KR6 R7002 robot. 
I would like to know details like urdf, mesh files and how I can interface KRC5Micro Controller With ROS.

gavanderhoorn, 2021-06-14:
I believe you mean the `KR 6 R700-2` variant, correct?

In any case: we don't have support for that specific variant yet, so you'd have to create the required files. It's likely you could start with the files for the non-`-2` in [kuka_kr6_support](https://github.com/ros-industrial/kuka_experimental/tree/984e1f214246ab7631d1727f5428b0c815fe3a34/kuka_kr6_support).

As to KR C5 support: I'm not sure whether there are any changes required to the RSI / EKI hw interfaces. You might need the changes proposed in #152.

simonschmeisser, 2021-09-28:
You can have a look at the `KR 10 R900-2`, it appears to be almost the same, only link 2 and 4 being shorter. There is an open PR #198 that you can have a look at for how to reuse most of an existing robot model. You would only need to add new meshes for link 2 and 4 in the `meshes` folder

Re KRC5: Kuka claims there are no incompatibilities. So if the code here works on KR4 it should run on KRC5 just as well.

longwoo, 2021-10-20:
need KR 6 R700-2 support too.