ros-industrial/kuka_experimental #193: Still active?

matteopantano, 2021-06-23:
I am planning to do some tests with this repo and I would like to connect directly to a kuka agilus. Is this something feasible? I am seeing that the latest upload was 4 yrs ago.

Best,
Matteo

gavanderhoorn, 2021-06-23:
RSI hasn't really changed, neither has EKI. So there isn't that much to do (other than improve UX and keep up with `ros_control` changes, which have been very few over the past few years, and we don't support ROS 2 here).

There are basically 3 versions of RSI actively used "in the wild": v2, v3 and v4.

v2 and v3 are supported directly. v4 is in #152 and #155.

gavanderhoorn, 2021-06-23:
As this is not an issue with the packages in this repository, I'm going to close it.

Feel free to keep commenting on it.

closed 2021-06-23 by gavanderhoorn

matteopantano, 2021-06-25:
Thanks a lot, this was the information I needed!