ros-industrial/kuka_experimental #194: Kuka model Random moves in Gazebo

ranjitkathiriya, 2021-07-28:
Hello there, 

I have implemented code from both branches https://github.com/tingelst/kuka_experimental/tree/add-kr16-gazebo, https://github.com/cschindlbeck/kuka_experimental/tree/gazebo_kr10 

When I am running my code in Gazebo I am getting random moves like the below youtube video. Any idea what can be wrong over here? Is it something wrong with gravity?

https://youtu.be/FgF1sTOMdsY

Thanks in advance for your help.

gavanderhoorn, 2021-09-21:
I believe [Kuka custom model trembling in Gazebo](https://answers.ros.org/question/385012) is a cross-post of this issue.

Having written that: I don't see 'trembling' in the video you link.

It looks like a Gazebo model for which the controllers are not active.

gavanderhoorn, 2021-09-21:
As this is not an issue with the packages in this repository, I'm going to close it.

Feel free to keep commenting on it, but perhaps your ROS Answers post would be a better place to discuss your problem.

closed 2021-09-21 by gavanderhoorn