ros-industrial/kuka_experimental #196: kuka_experimental/kuka_eki_hw_interface for KRC2

Levaru, 2021-09-20:
Hi,
we have loaned a robot to test our software but we were only able to get an old one with the KRC2 controller.
I successfully installed the KUKA Ethernet KRL Interface software (Version 1.2) but encountered problems when trying to implement kuka_experimental/kuka_eki_hw_interface.

It seems that the kuka_eki_hw_interface was written with KUKA.Ethernet KRL 2.x which has a different xml structure and commands compared to 1.2.

Here are some things that are different in 1.2:

- functions don't return eki_status but rather an integer
- function names begin with EKX instead of EKI
- EKI_INIT, EKI_Clear and EKI_CheckBuffer is missing
- The structure of EkiHwInterface.xml is divided into EkiHwInterface.xml, EkiHwInterface+.xml and XmlApiConfig.xml

I'm really lost concerning the last point. I was able to rename most of the functions but I'm not sure how to write those three files. I'm guessing that I'll also have to rewrite some of the source code for the kuka_eki_hw_interface ROS-node to make it compatible with the old version.

Does someone have any pointers on how to properly adjust the files for EKI V1.2?

gavanderhoorn, 2021-09-20:
I'm (personally) not aware of anyone having performed this migration/port.

You could see whether any of the forks contain the changes you need.

Levaru, 2021-09-28:
I managed to port it but it required some workarounds. The two most important changes are:

1. Changing the socket in the ROS-Program from a UDP-Client to a TCP-Server. I'm actually not sure if TCP is absolutely required but that's what the documentation says.
2. The 1.2 version doesn't have the `EKI_CheckBuffer()` function, so I had to implement a command counter that gets updated once the roboter reads the next command. Here I'm also not sure if this is the correct solution, because I get some very choppy/laggy movement.

Another thing that really confuses me are the `EKX_Get***Element()` functions. Parameter 1 of these functions has the following description:
> INTEGER
> Determines whether the next value or the current 
> value is to be read from the buffer
> INT=0: the next value is read (First In First Out) 
> INT=1: the current value is read (Last In First Out)

At first I had this set to 0 since I wouldn't want to miss any commands from the buffer right? This led to some strange movement behaviour, where the robot would oscillate during the movement by moving forward and then back a little bit, before executing the rest of the command normally. Some other times the robot would move 80% of the trajectory, then got back to the first 20%, then again forward to 80% and would repeat this multiple times, before finally reaching the target. After setting the parameter to 1 I didn't encounter this anymore, aside from some very choppy/laggy movement.

Here are my changes, could somebody please make a sanity check? 

EkiHwInterface.xml
```
<Elements>
  <Element Tag="RobotCommand" Type="STRUCTTAG" Stacksize="5" />
    <Element Tag="RobotCommand.ID" Type="INTEGER" Stacksize="5" />
    <Element Tag="RobotCommand.Pos" Type="STRUCTTAG" Stacksize="5" />
      <Element Tag="RobotCommand.Pos.A1" Type="REAL" Stacksize="5" />
      <Element Tag="RobotCommand.Pos.A2" Type="REAL" Stacksize="5" />
      <Element Tag="RobotCommand.Pos.A3" Type="REAL" Stacksize="5" />
      <Element Tag="RobotCommand.Pos.A4" Type="REAL" Stacksize="5" />
      <Element Tag="RobotCommand.Pos.A5" Type="REAL" Stacksize="5" />
      <Element Tag="RobotCommand.Pos.A6" Type="REAL" Stacksize="5" />
</Elements>
```

EkiHwInterface+.xml
```
<RobotState>
  <Pos A1="" A2="" A3="" A4="" A5="" A6="">
  </Pos>
  <Vel A1="" A2="" A3="" A4="" A5="" A6="">
  </Vel>
  <Eff A1="" A2="" A3="" A4="" A5="" A6="">
  </Eff>
  <RobotCommand ID="" Size="5">
  </RobotCommand>
</RobotState>
```

XmlApiConfig.xml
```
<?xml version="1.0"?>
<XmlApiConfig xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="XMLCommunicationSetup.xsd">
  <XmlApiParam InitOnce="false"/>
  <Channel SensorName="EkiHwInterface" SensorType="EkiHwInterface">
    <TCP_IP IP="192.168.16.98" Port="54601" Route="true" MapPort="54601"/>
  </Channel>
</XmlApiConfig>
```

kuka_eki_hw_interface.src
```
&ACCESS RVP
&REL 5
def  kuka_eki_hw_interface()
  ; Declarations
  decl axis joint_pos_tgt
  decl int elements_read

  ; INI
  bas(#initmov, 0)  ; Basic initialization of axes

  eki_hw_iface_init()
  
[... 256 lines cut ...]
        eki_acceptor_.accept(eki_server_socket_, eki_server_endpoint_);
        
        eki_acceptor_.close();

        // Start persistent actor to check for eki_read_state timeouts
        deadline_.expires_at(boost::posix_time::pos_infin);  // do nothing unit a read is invoked (deadline_ = +inf)
        eki_check_read_state_deadline();

        // Initialize joint_position_command_ from initial robot state (avoid bad (null) commands before controllers come up)
        if (!eki_read_state(joint_position_, joint_velocity_, joint_effort_, eki_cmd_buff_len_))
        {
            std::string msg = "Failed to read from robot EKI server within alloted time of "
                              + std::to_string(eki_read_state_timeout_) + " seconds.  Make sure eki_hw_interface is running "
                              "on the robot controller and all configurations are correct.";
            ROS_ERROR_STREAM(msg);
            throw std::runtime_error(msg);
        }
        joint_position_command_ = joint_position_;

        ROS_INFO_NAMED("kuka_eki_hw_interface", "... done. EKI hardware interface started!");
    }

} // namespace kuka_eki_hw_interface

```

gavanderhoorn, 2021-09-28:
Thanks for reporting back.

If you want to make it easier for others to check out your changes, please fork this repository, commit your changes to a new branch, then [open a Pull Request](https://docs.github.com/en/github/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request).

Right now it's really difficult to figure out what you've changed. By opening a PR, we immediately get a nicely rendered diff.

Levaru, 2021-09-28:
I wanted to avoid this, since I changed the indentation in all files. But I changed it all back and made a [pull-request](https://github.com/ros-industrial/kuka_experimental/pull/205) with a readable diff. I hope that I did this correctly.

Levaru, 2021-09-28:
Some additional info about the issue with the `EKX_Get***Element()` FIFO/LIFO parameter. Here is an image showing the state of a joint angle during each `eki_read_state` and the current `joint_state_command_` during that state. You can clearly see how the joint starts to move from 90deg to 0deg but about 87deg it moves back up again. This was with the parameter set to 0 which should be FIFO.
[image]

Levaru, 2021-09-28:
And this is the same movement but with the parameter set to 1. (It's also a little bit faster but that's not relevant)
[image]