ros-industrial/kuka_experimental #200: indigo compatibility

simonschmeisser, 2021-09-23:
I looked at the xacro wiki a bit yesterday and it seems that python functions are only supported starting with jade. We have been merging `radians(90)` and similar since at least 2018 it seems. Apparently some of the jade features of xacro can be enabled with `xacro --inorder`, which we don't use in the tests however.

Tests (failing) for indigo can be found here: https://github.com/simonschmeisser/kuka_experimental/runs/3679564585?check_suite_focus=true

I propose to add a new branch `melodic-devel` and continue updating this repo there. `kinetic-devel` would also be fine for me but melodic preferred.

Opinions @gavanderhoorn ?

gavanderhoorn, 2021-09-23:
> I propose to add a new branch `melodic-devel` and continue updating this repo there.

sounds good.

Don't forget to re-target outstanding PRs.

simonschmeisser, 2021-09-23:
Done. I seem to lack permission for changing the default branch, do we want to do that immediately and if so could you please do it?

simonschmeisser, 2021-09-23:
And re-enable branch protection (?)

closed 2021-09-28 by simonschmeisser