ros-industrial/kuka_experimental #201: RSI KSS29002 Error

lijianhualeigua, 2021-09-26:
Hi everyone,
We're using a kr6 r900-2, with KRC4 Compact, KSS 8.5.8 and RSI 4.0.9.
The RSI mode is  that: RSI_ON(#RELATIVE,#IPO,#BASE).
The TimeOut in the RSI object Ethernet is set to 10000.

Our system works with auto_ext, where the main computer controls which krl programs runs each time.

The rsi diagram did not contains a stop block which command to exit the RSI_MOVECORR mode. This stop block is triggered from a channel in the Ethernet block.

Usually, everything works perfectly fine.But if we do not let the robotic arm move in a period of time , usually a few minutes. The teach pendant will prompt the error of KSS 29002: Error Message: "Signal flow (running): object ETHERNETx return error RSITimeout".

But in this time, RSI is always connected,  it has been sending [0, 0, 0, 0, 0, 0]  because of RSI is use the relative mode, use for responding to RSI messages. The program pointer in the teach pendant Keep pointing at the RSI_MOVECORR() statement.

What does the RSIBad means? I have been always responding  the RSI message, why does it still prompt the kss29002 error of TImeOut?

I would appreciate any help.

Thanks!

gavanderhoorn, 2021-09-28:
Unless someone here has seen this specific error, I don't believe we can help you with this.

Have you tried asking KUKA, or posting on a robotics forum with a focus on KUKA specifically?

gavanderhoorn, 2022-03-10:
Closing due to inactivity.

closed 2022-03-10 by gavanderhoorn