ros-industrial/kuka_experimental #209: KUKA KR 4 R600 - ROS Support

koikonomou, 2021-11-18:
Hello,

I would like to know if there are any ROS controllers, urdf files and meshes for KR 4 R600.
Are there any instructions on how to use KR C5 micro controller with ROS ?

gavanderhoorn, 2021-11-19:
> I would like to know if there are any ROS controllers, urdf files and meshes for KR 4 R600.

All robot series/variants which are supported by this repository can be found in this repository. If you don't find something here, it may be supported somewhere else, but that would mean it's not supported here.

> Are there any instructions on how to use KR C5 micro controller with ROS ?

I haven't used that controller series yet myself, but IIRC, they still run KSS and are bw compatible with what we use here.

gavanderhoorn, 2021-11-19:
And just to clarify: "support" typically means two things:

 1. there is a robot support package (ie: `xacro:macro`, meshes, `.launch` files)
 2. the driver is compatible

Both drivers (RSI and EKI) are manipulator-agnostic. So support in this case would come down to a robot support package.

simonschmeisser, 2021-11-19:
Hi,

please see our version of `kuka_experimental` for a support package: https://github.com/isys-vision/kuka_experimental

We were told by Kuka that KRC 5 is fully compatible to KRC4 software wise (its KSS 8.7 vs KSS 8.6 as the software/operating system). Should receive one shortly to actually confirm this

gavanderhoorn, 2022-03-10:
@simonschmeisser wrote:
> please see our version of `kuka_experimental` for a support package

would you want to upstream that @simonschmeisser ?

simonschmeisser, 2022-03-10:
> Edit: o wait, you already have #210 open.
> 
> Is that PR in a reviewable state?

You always tend to find something :wink:  but after a quick glance I'd say I left it in a reviewable state, yes