ros-industrial/kuka_experimental #212: ROS support for Kuka KR8 with KR C5 controller

christianlandgraf, 2022-01-26:
Hi,

we are thinking of using ROS in an industrial project which requires a Kuka KR8 robot with a KR C5 control.
Does anyone have experience with such a setup and how robustly the ROS driver works?
We are planning to use it for a path following application. A working velocity controller might be sufficient, but some more advanced trajectory control would make it easier, I guess...

Thanks a lot!
Christian

simonschmeisser, 2022-01-26:
Hi Christian,

from our experience KRC5 is indeed fully compatible with KRC4 as claimed by Kuka. Our company (Optonic) uses an inhouse driver that uploads complete trajectories using EthernetKRL and executes them using the SPLINE commands. I plan on testing the two drivers included here (PTP via EthernetKRL or realtime via RSI) but Kuka has long delivery times currently ...

Adding a KR8 model based on the existing models is quite simple and we always welcome PRs!

You're welcome
Simon

christianlandgraf, 2022-01-26:
Thanks for the quick answers 👍 I thought it depends on the (brand-specific) hardware interface which `ros_control` components can be used... But I'm not an expert in this area, sorry if the question was misleading...

gavanderhoorn, 2022-01-26:
Nothing to apologise for.

> I thought it depends on the (brand-specific) hardware interface which ros_control components can be used..

no, not really.

What is important is which interfaces are exposed by the `RobotHW`. So if you have a controller which requires a velocity interface, but the target `RobotHW` does not expose those, it will obviously not work.

That's why I wrote:

> by default, both drivers only use absolute mode, so 'velocity control' is not supported.

What you do see sometimes though is that OEMs write custom interfaces (ie: outside the set of `ros_control` default ones). If you then write controllers which require *those* interfaces, you're very likely to not be able to reuse those controllers anywhere else.

One example of this would be the `RobotHW` provided by the `franka_hw` package. But there are certainly others.

None of the drivers in this repository use any custom interfaces, so that would not be a problem.

What could be a problem is the fact there is no support for velocity interfaces (as only absolute RSI and EKI are used).

gavanderhoorn, 2022-01-26:
re: compatibility: RSI versions have certainly changed (as they're sort-of tied to KSS versions), so that may require some updates to certain files.

There are a couple of open PRs addressing that.

gavanderhoorn, 2022-03-10:
I'm closing this as I believe we've addressed the original question.

It would be nice to know whether you got things to work in the end @christianlandgraf.

closed 2022-03-10 by gavanderhoorn