ros-industrial/kuka_experimental #213: Error trying to rostopic echo the state of the robot

TONiiV, 2022-02-08:
Hi,

I launched a kuka robot from the kuka_experimental using 

> roslaunch kuka_kr16_support load_kr16_2.launch

and then I tried to launch the kuka hw interface 

> roslaunch kuka_rsi_hw_interface test_hardware_interface.launch

It came up with some warnings, but it seems to connect to the robot successfully. 

Then when I try to 

> rostopic echo /position_trajectory_controller/state

this error poped up:

```
**Traceback (most recent call last):
  File "/home/ipa325/catkin_ws/src/kuka_experimental/kuka_rsi_simulator/scripts/kuka_rsi_simulator", line 80, in <module>
    rsi_act_pub.publish(msg)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 882, in publish
    self.impl.publish(data)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 1066, in publish
    serialize_message(b, self.seq, message)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/msg.py", line 152, in serialize_message
    msg.serialize(b)
  File "/opt/ros/noetic/lib/python3/dist-packages/std_msgs/msg/_String.py", line 56, in serialize
    _x = _x.encode('utf-8')
AttributeError: 'bytes' object has no attribute 'encode'**
```

It seems to be something wrong with python. The python version is 3.10, ros noetic and ubuntu LTS 20.14. Did anyone come across with the similar problem? **And btw is anyone using any Cartesian controller**? 

Thanks.

gavanderhoorn, 2022-02-08:
This doesn't look like an issue with one of the packages here.

`/position_trajectory_controller/state` is of type `control_msgs/JointTrajectoryControllerState`, which is a standard message from `control_msgs.

I'm going to close this, as I don't believe this is something we control / cause.

gavanderhoorn, 2022-02-08:
Having written that:

> The python version is 3.10

ROS Noetic is built against Python `3.8`, which is the default version of Python on Ubuntu Focal.

You may want to check what's going on there.

Especially around encoding/decoding of strings and bytearrays there have been a few breaking changes in various Python `3.x` versions.

I would not assume there is always (bw) compatibility between Python `3.8` and `3.10`.

gavanderhoorn, 2022-02-08:
Spoke too soon.

It could be something needs to be updated here:

https://github.com/ros-industrial/kuka_experimental/blob/bd165a723b0547f6daef4203bc8d0ef89188fae0/kuka_rsi_simulator/scripts/kuka_rsi_simulator#L13-L27

I can't test anything right now, but if `xml.etree.ElementTree.tostring()` returns a `bytes` instead of a `str` (or the other way around), `rospy.Publisher.publish(..)` will probably not be able to deal with that.

gavanderhoorn, 2022-02-08:
I'm confused why this would be happening when you `rostopic echo /position_trajectory_controller/state` btw.

The code mentioned in the trace has nothing to do with that topic.

gavanderhoorn, 2022-03-11:
Closing due to inactivity.

closed 2022-03-11 by gavanderhoorn