ros-industrial/kuka_experimental #215: How to simulate a full virtual controller with RSI?

aatb-ch, 2022-03-10:
I am finding very little information on the possibilities to simulate a virtual KRC4 controller driven over RSI, as I would be able to validate the protocol to install KRL files and configure network settings before having access to a real robot, which is not possible with the "simple" RSI simulator node provided here.

Is there a known way to do that? I downloaded Kuka.Sim, and checked about OfficeLite. Unlike RobotStudio for ABB with EGM I cannot find any information online and forums for Kuka and RSI. Wondering if anybody has successfully done it?

gavanderhoorn, 2022-03-10:
As this is not an issue with the packages in this repository, I'm going to close it.

closed 2022-03-10 by gavanderhoorn

aatb-ch, 2022-03-10:
Agreed this is not an issue with the package per-say, it seemed to me that it could have been a common use-case but I could not find anything in the past issues. Thanks I did not find this one on robot-forum, found some other RSI-related things but without any clear solution. Will investigate further this then.

simonschmeisser, 2022-03-10:
We use OfficeLite in combination with EthernetKRL. It's useful for syntax checking and ensuring basic functionality but way to slow for anything more serious. Note also that you have to pay for everything as if it where a real robot.

aatb-ch, 2022-03-10:
Thanks, what do you mean by "it's way too slow"? The VM itself is running too slow for something like RSI to not cause timeout issues? or EthernetKRL is slow? That would be expected since they are completely different approaches to online control no?

Yes this is indeed an problem regarding licensing costs, I had counted on the kuka.Sim trial to get me up to speed before jumping on the real robot but seems like OfficeLite is the way to go, but I also now realize that the licensing cost might be why there are so little people willing to get it to simulate instead of going on the real robot in T1.