ros-industrial/kuka_experimental #218: KUKA KR10 support for ignition gazebo and ros2.

addy1997, 2022-04-14:
Hello there,

I am working on an ignition gazebo to develop a package for KUKA robots to be supported by ROS2 and ignition gazebo. Currently, these files (especially, the launch files) are written to support ROS1. How do I change it for ROS2?

Thanks

gavanderhoorn, 2022-04-27:
You would have to migrate the packages and their contents to their ROS 2 counterparts.

We don't have specific instructions for that here.

I'd suggest to find existing material "on the internet", or specifically refer to the Ignition (now called *Gazebo* again) documentation/tutorials.

gavanderhoorn, 2022-04-27:
I'm going to close this, as it's not (really) an issue with the packages in this repository.

Feel free to keep commenting of course.

closed 2022-04-27 by gavanderhoorn