ros-industrial/kuka_experimental #221: Missing KR20-R3100

danzeeeman, 2022-06-13:


simonschmeisser, 2022-06-19:
Thanks for leading me to https://github.com/openframeworks/openFrameworks which looks quite interesting but generally this is a community driven effort here so we highly welcome PRs to contribute this model as well. Some models of Kuka robots share kinematics and shape but trade payload (the `20` part) for speed. In this case it would be especially easy to contribute

closed 2022-06-19 by simonschmeisser

danzeeeman, 2022-06-19:
Why are you closing this I don’t have the URDF….it should remain open until
i or someone else fixes it.

This bot is the larger 3100 mm reach so it doesn’t share much with the
other bots in the repo…the fact that it is missing is an issue so please
reopen this bug

gavanderhoorn, 2022-06-20:
I assume @simonschmeisser closed it because there are *many* models/variants 'missing', and we don't track those either with issues.

A model not having a support package isn't exactly a `bug`, it's expected: unless/until someone contributes a support package for a specific variant, there just won't be any support.

We reserve `bug` for issues which report behaviour or models which is/are incorrect, and need fixing. By definition, something which doesn't exist can't really exhibit bugs then.

I'm willing to re-open, but that would make more sense if you would be willing to address the issue by contributing support for the variant.