ros-industrial/kuka_experimental #234: External instruction rsi_krlExterXML not load

Prawee682, 2023-03-07:
Hello, I'm a newbie in using kuka and ros together, I'm using kuka kr16-2 and I'm having the following problem.

On PC when I run terminal

`roslaunch kuka_rsi_hw_interface test_hardware_interface.launch sim:=false`

Everything was fine, and I put `ros_rsi.src` to `C:\KRC\ROBOTER\KRC\R1\Program`, and `ros_rsi_ethernet.xml` to `C:\KRC\ROBOTER\Init` by set port and ip_address successfully. When I press run button `ros_rsi.src` on teach pad, I reached the step where on the pad it says "`Programmed path reached (BCO)`" and my robot returns to home, but when I press the run button again I get

`External instruction rsi_krlExterXML not load`

Which as I read in the README.md it should get "`Attention –RSI sensor mode active`" right? 

How should I fix it?

tingelst, 2023-03-07:
The rest of the files should be copied to: C:\KRC\ROBOTER\Config\User\Common\SensorInterface

Please follow the instuctions found [here ](https://github.com/ros-industrial/kuka_experimental/blob/melodic-devel/kuka_rsi_hw_interface/krl/KR_C4/README.md).

Prawee682, 2023-03-07:
> The rest of the files should be copied to: C:\KRC\ROBOTER\Config\User\Common\SensorInterface
> 
> Please follow the instructions found [here ](https://github.com/ros-industrial/kuka_experimental/blob/melodic-devel/kuka_rsi_hw_interface/krl/KR_C4/README.md).

Ok, I will try. Thank you very much.